patrickcollins12/esphome-fan-controller #2: cool setup just what i was looking for but my fan does not shut off totally

dartfrogdk, 2022-06-13:
i set the fan at 0% and 50% but still it wont stop running totally, do you know what i can do

sophof, 2022-06-20:
I was working on the electrical setup of my own implementation of this this weekend and ran into the same problem, apparently some fans are just set up this way. 
You have to use a relay to switch off the power entirely to the fan. I needed a level shifter as well to both up the PWM signal and the relay control signal to 5v. The next hurdle was that the fan then would run purely from the pwm output, so I also needed to switch of the level shifter output, luckily my level shifter allowed for control there too.

The end result (just for the fan for now, no PID control yet) looks as follows in the yaml:
``` yaml
output:
  - platform: gpio
    pin: GPIO26
    id: pwm_on
    inverted: true
  - platform: gpio
    pin: GPIO27
    id: fan_supply
  - platform: ledc
    pin: GPIO14
    frequency: 25000 Hz
    id: testfan

fan:
  - platform: speed
    output: testfan
    name: "Test Fan"
    on_turn_on:
      - output.turn_on: fan_supply
      - logger.log: "Power of Fan turned ON"
      - output.turn_on: pwm_on
      - logger.log: "PWM enabled"
    on_turn_off:
      - output.turn_off: fan_supply
      - logger.log: "Power of Fan turned OFF"
      - output.turn_off: pwm_on
      - logger.log: "PWM disabled"
```
With this the fan switches off at 0. 1% is then the startup RPM of my fan.

closed 2022-09-08 by patrickcollins12

sophof, 2023-04-05:
It is hard to parse your comment this way, can you put your yaml in a code block? Just surround it with ``` before and after the code.

I changed my code a lot in the mean time and right now I have a much simpler version, where the fans are always on if my receiver is on in the TV cabinet that I'm cooling. Since I'm using two 140mm fans I literally don't hear them anyway on the lowest speed.

It simply imports a binary sensor from home assistant to check if the receiver is on, and if it is, switches the climate to COOL and turns on the relay and PWM. I also added a few simple buttons to be able to do it by hand (but I never have to use them anymore).
My code now looks as follows:
```
substitutions:
  friendly_name: Receiver Fan

esphome:
  name: tv-meubel-fan-links

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  fast_connect: true

dallas:
  - pin: 26
    update_interval: 10s
 
binary_sensor:
  - platform: homeassistant
    id: receiver_power
    entity_id: binary_sensor.receiver_power
    on_state:
      then:
        - script.execute: toggle_climate
        
script:
  - id: toggle_climate
    mode: restart 
    then:
      if:
        condition:
          binary_sensor.is_on: receiver_power
        then:
          - climate.control:
              id: receiver_thermostat
              mode: "COOL"
        else:
          - delay: 5 min
          - climate.control:
              id: receiver_thermostat
              mode: "OFF"

sensor:
# ########################################################
# # START THE FAN CONTROLLER SETUP

  - platform: template
    name: $friendly_name p term
    id: p_term
    unit_of_measurement: "%"
    accuracy_decimals: 2

  - platform: template
    name: $friendly_name i term
    id: i_term
    unit_of_measurement: "%"
    accuracy_decimals: 2

  - platform: template
    name: $friendly_name d term
    id: d_term
    unit_of_measurement: "%"
    accuracy_decimals: 2

  - platform: template
    name: $friendly_name output value
    unit_of_measurement: "%"
    id: o_term
    accuracy_decimals: 2

  - platform: template
    name: $friendly_name error value
    id: e_term
    accuracy_decimals: 2
      
  - platform: dallas
    address: 0x940000000ca14e28
    name: "Receiver Temperature"
    id: receiver_temperature
    accuracy_decimals: 1
    filters:
    - filter_out: 85.0
    - throttle: 30s

  # Take the "COOL" value of the pid and send 
  # it to the frontend to graph the output voltage
  - platform: pid
    name: "Fan Speed (PWM Voltage)"
    climate_id: receiver_thermostat
    type: COOL

output:
  - platform: ledc
    id: fan_speed
    pin: GPIO19
    frequency: "25000 Hz"
    
switch:
  - platform: gpio
    name: PWM gpio
    pin:
      number: GPIO18
      inverted: true
    id: pwm_on
  - platform: gpio
    name: Fan Power
    pin:
      number: GPIO03
    id: fan_supply

climate:
  - platform: pid
    name: "Receiver Fan Thermostat"
    id: receiver_thermostat
    sensor: receiver_temperature

    default_target_temperature: 33°C
    cool_output: fan_speed

    deadband_parameters:
      threshold_high: 0.4°C
      threshold_low: -1.0°C
      kp_multiplier: 0.0   # proportional gain turned off inside deadband
      ki_multiplier: 0.2  # integral accumulates at only 20% of normal ki
      kd_multiplier: 0.0   # derviative is turned off inside deadband
      deadband_output_averaging_samples: 10   # average the output over 10 samples within the deadband

    on_state:
      - sensor.template.publish:
          id: p_term
          state: !lambda 'return -id(receiver_thermostat).get_proportional_term() * 100.0;'
      - sensor.template.publish:
          id: i_term
          state: !lambda 'return -id(receiver_thermostat).get_integral_term()* 100.0;'
      - sensor.template.publish:
          id: d_term
          state: !lambda 'return -id(receiver_thermostat).get_derivative_term()* 100.0;'
      - sensor.template.publish:
          id: o_term
          state: !lambda 'return -id(receiver_thermostat).get_output_value()* 100.0;'
      - sensor.template.publish:
          id: e_term
          state: !lambda 'return -id(receiver_thermostat).get_error_value();'
      - if:
          condition:
            and:
              - lambda: 'return id(receiver_thermostat).mode == esphome::climate::ClimateMode::CLIMATE_MODE_OFF;'
              - switch.is_on: fan_supply
              - switch.is_on: pwm_on
          then:
            - switch.turn_off: fan_supply
            - switch.turn_off: pwm_on
            - logger.log: "Fan switched off"
      - if:
          condition:
            and:
              - lambda: 'return id(receiver_thermostat).mode != esphome::climate::ClimateMode::CLIMATE_MODE_OFF;'
              - switch.is_off: fan_supply
              - switch.is_off: pwm_on
          then:
            - switch.turn_on: fan_supply
            - switch.turn_on: pwm_on
            - logger.log: "Fan switched on"

    # The extents of the HA Thermostat
    visual:
      min_temperature: 20 °C
      max_temperature: 50 °C
      temperature_step: 1 °C
  
    # See the README for setting up these parameters.
    # These are over ridden by the number templates above.
    control_parameters:
      kp: 0.05
      ki: 0.0002
      kd: 0
      max_integral: 0.0
      output_averaging_samples: 4
```