patrickcollins12/esphome-fan-controller #5: [Request] Possible to use two temperature sensors?

martijn-brant, 2022-07-07:
Thanks for this repo.. the code and schematics work great for me. My prototype is working great however I have a question.. is it possible to add another DHT temperature probe? My AV cabinets have two zones and I've added a fan in each. One is for the AV Receiver (always on when using TV) and the other is for gaming consoles (only on sometimes). I'd rather not build two separate fan controllers for two compartments. I don't mind both sides spinning faster when only one zone is needing more cooling (e.g. the gaming side being cooled even if there are no consoles active).

I had to reassign the DHT to GPIO23 on my nodemcu-32s board else it wouldn't work. PWM is on GPIO15. I've looking at the code as to what would be needed to support a 2nd DHT but the code is a bit too advanced for me. Ideally it would report two temperature values to Home Assistant as well and act on the worst one. But I would be more than pleased to get the worst value of the two as well. (one being 29c and the other 35c.. having it act like it's 35c and adjust fan accordingly).

Could you help with this? Many thanks.

patrickcollins12, 2022-07-08:
glad it worked for you!

Well you could easily run two separate systems off the one esp32: two pids, two dht, two fans. You would just copy all of those sections in the yaml file and give each a different name.

However, to answer your question directly. Yes you can take the max of two sensors to create a 3rd ”template” sensor. All three sensors will be visible to home assistant.

I’m not at my computer at the moment, so I can’t provide direct code.

This example creates a 3rd template sensor which is an average.

https://community.home-assistant.io/t/average-temperature-based-on-multiple-sensors/220781/3

to create a max instead the lambda is something like:

```yaml
lambda: |-
      return max(id(dht_temp_1).state, id(dht_temp_2).state);
```

let me know how you go.

martijn-brant, 2022-07-08:
Ah yes, great idea just to duplicate the YAML values for the sensors/outputs. I added an additional GPIO for extra PWM (Fan 2) and extra DHT11. Works like a charm. This gives me dual-zone option on a single Nodemcu. Thanks for your code and your reply.

closed 2022-07-08 by martijn-brant

B08Z, 2023-06-18:
> Ah yes, great idea just to duplicate the YAML values for the sensors/outputs. I added an additional GPIO for extra PWM (Fan 2) and extra DHT11. Works like a charm. This gives me dual-zone option on a single Nodemcu. Thanks for your code and your reply.

Hello,

Could you share you working yaml. I am about to start and may end up with 3 fans as they are a little small ( not ideal I know but got to work with what you got.)

Anyway if you could share how you did dual fans and dual dht temp that would be great

bartaspoz, 2023-07-22:
> Ah yes, great idea just to duplicate the YAML values for the sensors/outputs. I added an additional GPIO for extra PWM (Fan 2) and extra DHT11. Works like a charm. This gives me dual-zone option on a single Nodemcu. Thanks for your code and your reply.

Hey! I would be also interested in the code supporting two zones :) Many thanks!

martijn-brant, 2023-07-22:
Ah sorry I missed your first message @B08Z .. @bartaspoz here is the code:

`#based on https://github.com/patrickcollins12/esphome-fan-controller
substitutions:
  friendly_name: Console Fan

esphome:
  name: console-fan

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Console-Fan Fallback Hotspot"
    password: "xxx"

captive_portal:

globals:
  - id: dhttemp
    type: float
    restore_value: yes
    initial_value: '0'

number:

  ## RECEIVE kp,ki and kd parameters from input_text.kx helpers in 
  # Home Assistant. See the PID controller below
  # These helper values will get saved to flash thus permanently over-riding 
  # the initial values set in the PID below.

  # KP
  - platform: template
    name: $friendly_name kp
    icon: mdi:chart-bell-curve
    restore_value: true
    min_value: 0
    max_value: 50
    step: 0.001
    set_action: 
      lambda: |- 
        // ESP_LOGI("main", "!!!!!! kp from boot %d", id("console_fan_kp") );
        // id(console_thermostat).set_kp( id("$friendly_name kp") );
        id(console_thermostat).set_kp( x );

  # KI
  - platform: template
    name: $friendly_name ki
    icon: mdi:chart-bell-curve
    restore_value: true
    min_value: 0
    max_value: 50
    step: 0.0001
    set_action: 
[... 316 lines cut ...]
          id: e_term
          state: !lambda 'return -id(console_thermostat2).get_error_value();'
        
    # dummy heater. enable this if using autotune
    # heat_output: console_heat_speed

    # The extents of the HA Thermostat
    visual:
      min_temperature: 20 °C
      max_temperature: 50 °C
  
    # See the README for setting up these parameters.
    # These are over ridden by the number templates above.
    control_parameters:
      kp: 0.15
      ki: 0.004
      kd: 0.5
      max_integral: 0.0
  

switch:
  # Expose an ESP32 restart button to HA
  - platform: restart
    name: "Console Fan ESP32 Restart"
`

Bond246, 2023-09-05:
Hello,

i want to hijack this closed thread because i'm trying to do the same. The first step i've done was to copy the DHT sensor yaml-part, pasted it down below and changed the IDs and names.

But whatever i do the 2nd sensor is always "NA%". No values. 
I've already tried different GPIO Inputs, used different sensors. Attached the working sensor to the 2nd GPIO and so much more.

What could be wrong?
Instead of the DHT11 i'm using DHT22 sensors for my project.

``` yaml
sensor:
  # GET TEMP/HUMIDITY FROM DHT22
  - platform: dht
    pin: GPIO18
    model: DHT22
    temperature:
      name: "Temperature Unten"
      id: console_fan_temperature_1
      accuracy_decimals: 3

      # If you don't smooth the temperature readings 
      # the PID controller over reacts to small changes.
      filters:
         - exponential_moving_average:  
             alpha: 0.1
             send_every: 1

    humidity:
      name: "Humidity Unten"
      id: console_fan_humidity_1

    # the DHT11 can only be read every 1s. Use 1.3s to be safe.
    update_interval: 1.3s
  
  # GET TEMP/HUMIDITY FROM DHT22
  - platform: dht
    pin: GPIO23
    model: DHT22
    temperature:
      name: "Temperature Oben"
      id: console_fan_temperature_2
      accuracy_decimals: 3

      # If you don't smooth the temperature readings 
      # the PID controller over reacts to small changes.
      filters:
         - exponential_moving_average:  
             alpha: 0.1
             send_every: 1

    humidity:
      name: "Humidity Oben"
      id: console_fan_humidity_2

    # the DHT11 can only be read every 1s. Use 1.3s to be safe.
    update_interval: 1.3s
```