patrickcollins12/esphome-fan-controller #7: No longer able to compile.

prokyle123, 2023-01-02:
I was trying to change some of the code and whenever i try to upload to the esp i get the following error.. it happens on 2 different boards. i also would like to make another one but cant if it wont upload. could you provide any help? i love this project and i get ALOT of use out of it.  

I think it happened after an update and i feel like a url or dependency is no longer available.im completely unsure,

Processing aquariumcooler (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 5.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript '/data/cache/platformio/packages/framework-arduinoespressif32/tools/platformio-build.py'
File "/data/cache/platformio/platforms/espressif32/builder/frameworks/arduino.py", line 41, in <module>
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing FS
Warning! Could not find the package with 'FS' requirements for your system 'linux_x86_64'
Library Manager: Installing Update
Warning! Could not find the package with 'Update' requirements for your system 'linux_x86_64'
Library Manager: Installing ESPmDNS
Warning! Could not find the package with 'ESPmDNS' requirements for your system 'linux_x86_64'
Dependency Graph
|-- AsyncTCP-esphome @ 1.2.2
|-- WiFi @ 1.2.7
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- AsyncTCP-esphome @ 1.2.2
|-- DNSServer @ 1.1.0
|-- noise-c @ 0.1.4
|   |-- libsodium @ 1.10018.1
Compiling /data/aquariumcooler/.pioenvs/aquariumcooler/src/esphome/components/adc/adc_sensor.cpp.o
Compiling /data/aquariumcooler/.pioenvs/aquariumcooler/src/esphome/components/api/api_connection.cpp.o
In file included from src/esphome/components/adc/adc_sensor.h:4,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
src/esphome/core/hal.h:14:10: fatal error: esp_attr.h: No such file or directory

******************************************************************
* Looking for esp_attr.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_attr.h"
* Web  > https://registry.platformio.org/search?q=header:esp_attr.h
*
******************************************************************

 #include <esp_attr.h>
          ^~~~~~~~~~~~
compilation terminated.
*** [/data/aquariumcooler/.pioenvs/aquariumcooler/src/esphome/components/adc/adc_sensor.cpp.o] Error 1
In file included from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_frame_helper.h:13,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/socket/headers.h:116:10: fatal error: sys/ioctl.h: No such file or directory
 #include <sys/ioctl.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [/data/aquariumcooler/.pioenvs/aquariumcooler/src/esphome/components/api/api_connection.cpp.o] Error 1
========================== [FAILED] Took 1.99 seconds ==========================

TheDK, 2023-01-15:
Just compiled it on ESPHome 2022.12.3 without a problem...

patrickcollins12, 2023-01-17:
Great to hear. I suspect it was a broken platformio or esphome

prokyle123, 2023-01-17:
i fixed my issue by using the DEPRECATED version of esphome. Thank you.

closed 2023-01-21 by patrickcollins12