patrickcollins12/esphome-fan-controller #8: Fan Override does not work

TheDK, 2023-01-15:
When setting the Fan to On (while disabling the thermostat) and setting a value the fans do spin up but go down again as soon as the next temperature reading comes in. So the override does not seem to be stable or I am doing it wrong?

patrickcollins12, 2023-01-17:
The PID controller is still running. Turn the thermostat off. 

You'll need to add a separate power switch to control the fan output manually.

operinko, 2023-02-03:
The answer to your last question depends heavily on the fan you use.
*SOME* fans turn off completely at low PWM, but my corsairs, for example, have a set minimum they'll spin at even with a zero PWM signal.

DunklesKaltesNichts, 2023-02-03:
> I guess a power switch would only be needed to turn off the fan completely (can PWM send "real 0" or would a really cutting the 12V+ be better?).

I recommend fans that stop at 0%.
My Arctic PWM PST spin at 0% with 200RpM, the Artic PWM PST 0dB stop completely.

TheDK, 2023-02-03:
That is understood. So my question remains: How to stop the PID controller? :) But it's not that important... ;)

AlpineWhite, 2023-02-03:
> 

Some fans will turn off with a 100% PWM signal. I think deepcool will treat 100% as 0. I can vouch that no corsair fan behaves that way and no noctua fan behaves that way. I am having difficulty finding it, but another repo with ledc pwm control shows a deepcool 140mm fan turning off at 100% PWM (on an 8266 using sw pwm at 25khz). 
Of course, you can also just add a FET to switch the 12v.

patrickcollins12, 2023-02-03:
@TheDK forgive me for pointing out the obvious, but this is how you turn off the pid controller 
[image]

I was not aware that some fans don't treat 0% as off. I think the documentation should be updated to mention this. 
As @AlpineWhite indicated, it needs a FET capable of switching from the ESP32. I'd recommend IRLB8721 but an old IRF540N will probably also do the job. Do we need to modify the documentation to explain how to do this?

TheDK, 2023-02-04:
That is understood, here is the behaviour I am seeing:

1. PID controller is running, all good.
2. I turn the thermostat to OFF as shown above
3. I activate manual control and set the fan to a certain setting --> the fan reacts.
4. After the next temp sensor reading the PID controller regulates the sensor back (even though the thermostat is off).

So, as far as I can tell, the PID controller runs on the last set target temperature regardless of the thermostat being set to COOL or OFF.

TheDK, 2023-02-06:
...and to add another question: When I try the change the number entities (i.e. number.serverrack_fan_kd) nothing happens in the HA UI, nothing seems to happen at the ESP32 and when I open the window again it stayed at the old value. It appears to me HA does not read the number and publish it back to the ESP?

DunklesKaltesNichts, 2023-02-08:
I had the same problem.
I replaced:
```

set_action: 
      lambda: |- 
        id(console_thermostat).set_kp( x )
```

with:

```
on_value:
      then:
       lambda: id(${nodename}_console_thermostat).set_kp( x );
```

sha0x404, 2023-05-14:
same errore. i need some help.

pacmac, 2023-06-16:
I have the same problem, I want to force the fan to stay on irregardless of the temperature.

When I turn the switch on, the fan starts on full speed as expected, but a few seconds later the climate / pid switches it off again.

Is there a software solution to this as it's already deployed:

```
switch:

  - platform: template
    name: Fan ON
    id: fan_on
    icon: mdi:fan
    optimistic: true
    turn_on_action:
      - climate.control:
          id: pid_data
          mode: "OFF"
      - output.set_level:
          id: console_fan_speed
          level: 1
          
    turn_off_action:
      - climate.control:
          id: pid_data
          mode: "COOL"
```

darmach, 2024-01-18:
I'm running into the same thing @patrickcollins12 - it appears that setting the climate component to `off` does not disable the PID running. 
1. I switch climate from `cool` to `off` in HA
2. Enable fan control
3. Fan starts spinning
4. With next data received from ESP fan stops spinning

patrickcollins12, 2024-01-19:
Can we start a new issue please?

patrickcollins12, 2024-01-22:
@darmach what do you mean by step 2, "enable fan control"?

darmach, 2024-01-22:
Hi @patrickcollins12 I can open a new issue, no problem :)
As for the no.2 - by "enabling fan control" I meant turning on/setting speed to maximum for fan control device - the one templated in:
```
# Good for debugging, you can manually set the fan
# speed. Just make sure the Climate device is set to off or it will keep getting overridden.
fan:
  - platform: speed
    output: av_cabinet_fan_speed
    name: "Fan Speed"
```
So as such, I suspect it is being overridden  after all - despite climate device being turned off (Set to OFF in homeassistant instead of COOL)

patrickcollins12, 2024-01-22:
Interesting I understand the issue now. I wonder if that is new behaviour in esphome. The template need some kind of awareness who is in control.

patrickcollins12, 2024-01-27:
I'm pleased to announce a fix to this behavior. 

I've just pushed a new version of the console-fan.yaml.

But here are the specific changes.

There is a new "fan" added called "manual_fan_control".

Both this fan and the pid controller will output their values to a proxy output. This proxy output decides what to do. Basically if the manual fan control is on, it uses the speed value from that fan, otherwise it uses the value from the PID output value.

Also if you look at the new dashboard yaml I posted in the README.md it describes how to conditionally prevent display of all the PID stuff if in manual control.

```yaml
  # Every time the fan speed is updated, this sensor will
  # also be updated for displaying on the frontend. 
  # See proxy_output.
  - platform: template
    name: "Fan Speed (PWM Voltage)"
    unit_of_measurement: "%"
    id: fan_speed_pwm_voltage

output:
  # Wire this pin (13) into the PWM pin of your 12v fan
  # ledc is the name of the pwm output system on an esp32
  - platform: ledc
    id: console_fan_speed
    pin: GPIO13

    # 25KHz is standard PC fan frequency, minimises buzzing
    frequency: "25000 Hz" 

    # my fans stop working below 13% powerful.
    # also they're  powerful and loud, cap their max speed to 80%
    min_power: 13%
    max_power: 80%

  # This proxy output takes its input
  # if the manual fan control is on, use the level from that
  # otherwise use the PID control value.
  # Then publish the result to the fan (ledc) and 
  # also publish to the template output sensor
  - platform: template
    id: proxy_output
    type: float
    write_action:
      lambda: |-
        float write_val = 
          (id(manual_fan_control).state) ?
            id(manual_fan_control).speed / 100.0:
            write_val = state*1.0;
        id(console_fan_speed).set_level(write_val);
        id(fan_speed_pwm_voltage).publish_state(write_val*100.0);

# If you turn this on, you can manually set the fan speed.
# The PID will be ignored. This is done via the proxy_output.
fan:
  - platform: speed
    id: manual_fan_control
    output: proxy_output
    name: "Manual Fan Speed"

# Expose a PID-controlled Thermostat
# Manual: https://esphome.io/components/climate/pid.html
climate:
  - platform: pid
    name: "Console Fan Thermostat"
    id: console_thermostat
    sensor: console_fan_temperature

    # It is summer right now, so 30c is a decent target.
    default_target_temperature: 30°C
    cool_output: proxy_output
    # cool_output: console_fan_speed
```

Enjoy!

/cc @TheDK @darmach @pacmac

patrickcollins12, 2024-01-27:
[image]