patrickcollins12/esphome-fan-controller #9: Cool project!

jazzmonger, 2023-01-17:
Nice job on this! I'm totally stealing your PID efforts for 2 of my projects.

The first (and easier) is for my solar battery heaters.  I just finished getting a dimmer circuit working for 4 heating pads. 
https://diysolarforum.com/threads/new-battery-heater-controller-home-assistant-and-esphome.53665/#post-685275
I just got the ESPHome file to compile with all your PID stuff and added all the sensors to home assistant.  It's late and I'll head to my shop in the AM to try and understand how this cool PID controller works. 

The second, and far more advanced use case, is for my Smart Wood Pellet Stove project. I finally have victory over this pellet stove after a year of slogging it to make it actually smart with 685 lines of ESPHOME code to control every aspect of it.
https://github.com/jazzmonger/wood-stove-with-TYWE1S-Tuya-chip

My plan is to employ the PID component to control the  Auger feed motor control for the pellets to barely keep the stove lit. You can see the fire burning and cooling and burning and cooling, etc, etc as pellets are added and then backed off. 

[image]

https://github.com/jazzmonger/wood-stove-with-TYWE1S-Tuya-chip/discussions/28

I'm pondering what params to play with to use for the PID control line. I have a few options:
Auger on time ( this is tricky as too little and it goes out too much and u get smoke)
Auger off time - how long before more pellets are added
Stoking temp - how low the exhaust temp goes before manually stoking the fire w more pellets

I you have any thoughts on this after the dozens of hours you probably spent on figuring out the PID stuff, I'm all ears!

patrickcollins12, 2023-01-17:
Hey there, I've taken over ownership of the pid component in esphome.

I'm about to buy my first smoker so I'm interested in for you go.

A few things to look into:

If the auger is simply on or off, take a look at slowpwm which you can use to emulate quarter or half speed. PID climate can use slowpwm natively.

The auger is only one way to control the heat. Another way is by opening and closing an exhaust vent to control air flow. You could use a stepper motor.

It will be slow to get pid parameters from the pid autotune feature but it might be the best way.

jazzmonger, 2023-01-17:
well then!  You better make sure everything is in order on that component!  One complaint I have about the ESPHome docs is that there aren't any links to examples of projects using a component. This would help a great deal!

I don't have any options for controlling venting unfortunately. The only control I have is the auger motor.

are you on Discord?

Jeff

patrickcollins12, 2023-01-22:
Discord: pcollins12

closed 2023-02-04 by patrickcollins12