patrickcollins12/esphome-fan-controller #12: Reduce flash writes?

TheDK, 2023-01-28:
As far as I can tell the config you have writes a couple of parameters to flash every minute, amounting to ~0.5m writes per year. From what I've read the flash memory in ESP32 is expected to last 100k-1m write operations. I don't know if there's wear leveling in an ESP (guess not), so it might make sense to reduce writing the parameters to the ESP every 15min or so?

DunklesKaltesNichts, 2023-01-29:
my ESP has been running for almost 2 years without any problems.

the variable DHTTemp is not used anyway... you could switch off "restore_value" for all variables and after you have found suitable PID parameters, you could write them once into the code and flash them.

or just set "flash_write_interval" to 15min

patrickcollins12, 2023-02-04:
OK this is a good point, thanks DK. I'll the documentation to mention this.

TheDK, 2023-02-04:
I've added it to the yaml together with a comment, so it is in by design and if someone wants to take it out it can just be deleted.

patrickcollins12, 2023-02-05:
great thanks, i merged this. appreciate it.

closed 2023-02-05 by patrickcollins12