patrickcollins12/esphome-fan-controller #18: One console with two separate connected fans

marxbenjamin, 2023-04-24:
Hello,

first of all respect and thanks for the guide! It was fun building it.
I connected two fans (one for in, one for out) to different pins. - I know I could have just soldered them together, but I didn't want to. - I can now control both separately and also get all sensor values output separately.

But I would like to specify only one target temperature value via the console and this should be given to both output IDs.

Is this possible? As far as I understand, I would have to specify both IDs of the "cool_output" here in the code. But I can't find the syntax for it.

```
climate:
  - platform: pid
    name: "Console Fan Thermostat Out"
    id: console_thermostat_out
    sensor: console_fan_temperature

    # It is summer right now, so 30c is a decent target.
    default_target_temperature: 30°C
    cool_output: console_fan_speed
```

DunklesKaltesNichts, 2023-04-24:
thats not possible.
but you can use my solution:

```
    id: ${nodename}_power_fans
# Merging GPIO26 and GPIO25, no hardware modification necessary    
  - platform: template
    id: pwm_merge
    type: float
    write_action:
      - output.set_level:
          id: ${nodename}_pwm_inward
          level: !lambda return state;
      - output.set_level:
          id: ${nodename}_pwm_outward
          level: !lambda return state;
# PWM Fan inward
  - platform: ledc
    pin: GPIO26
    id: ${nodename}_pwm_inward
    frequency: 25khz
# PWM fan  outward
  - platform: ledc
    pin: GPIO25
    id: ${nodename}_pwm_outward
    frequency: 25khz 
```

marxbenjamin, 2023-04-25:
Thanks very much! It works as expected! :)

closed 2023-04-25 by marxbenjamin