patrickcollins12/esphome-fan-controller #19: LED strip control

smathev, 2023-04-26:
Hi.

Really great project! It seems to be exactly what I'm looking for. I found it while searching for ways to make a 3D printer enclosure, like this: https://www.reddit.com/r/3Dprinting/comments/10io7st/how_the_ikea_platsa_3d_printer_enclosure_looks/

So, in the enclosure I would use your project to add a bit of active cooling / stabilize temperature. However, I would also like to add some lights in there - like the linked enclosure. 

So, I figured the ESPHome would be able to control the lights, but to simplify the project I wanted to avoid also needing a 5V rail - So, I found these LED-strips: WS2815.

I was wondering if you knew whether it would be as simply as adding them to the 12V rail and adding them in the ESPHome component?

I realize this project isn't about lighting via ESPHome and it might not be needed in most peoples "media console housing", but I figured I would ask here since I consider my idea/thought an extension of your project.

Thank you for any input you could provide :)

patrickcollins12, 2023-04-29:
So I had the same need for my 3d Printer. If you're using Octoprint, I ended up using OctoRelay to plug a 12v light into the raspberry pi. The advantage of this is that you can control the light automatically when the 3d printer is running. To power the 12v lights I needed to use a mosfet.

That all being said, yes you could easily do the same from the ESP32:
ESPhome > ESP32 GPIO Out > Mosfet > 12v LEDs.

smathev, 2023-05-01:
Oh, ok. 

So, I have no idea how I would add a Mosfest into this. I was looking at your schematics and thinking I just needed to run the 12V power to the LED strip, a ESP32 GPIO OUT to the controller of the 12V strip and the ground to a ground on the ESP32.

But, I'm guessing that's wrong from your description :)

So, I would need this: https://www.amazon.de/-/en/MOSFET-Switching-Voltage-Control-Transistor/dp/B08CDXCL83

And then connect the 12V and a GPIO out from the Arduino to the MOSFET and further on to this: https://ardustore.dk/produkt/ws2815-5050-rgb-led-strip-12v-dc
I'm really just guessing here.

 Would it be too big an ask for you to add your suggestion into the very clear schematic you have on your README from this product?

Again, I understand if this is way out of line, and I realize it's not part of your project at all. But any additional help would be appreciated :)

closed 2023-06-02 by patrickcollins12