patrickcollins12/esphome-fan-controller #20: Set the target temperature from an automation

marxbenjamin, 2023-04-29:
I installed your fan-project in my NAS cabinet with two compartments. I have the temperature sensor at the top, because that's where my NAS is, which runs during the day. Below I have my backup NAS and my Raspberry.
I know, I just can install a second thermostat and use it. But I just want to monitore the system and HDD temperatures and want to set the PWM to ~100% or the target temperature to 20° just in case of an emergency. I don't think I'll need it, but having is better than needing.
So I wanted to ask if I can change the PWM. Since it would probably be constantly overwritten by the controller, lowering the target temperature is probably a better choice. 
However, I couldn't find it. Is there a way to do this that I've just overlooked?

Thanks in advance!

marxbenjamin, 2023-04-29:
Sorry...I kept trying and finally found it!

[image]

closed 2023-04-29 by marxbenjamin

patrickcollins12, 2023-04-29:
That strategy works. The other idea is turn off the climate device and manually set the fan speed. You can do that in the YAML.

marxbenjamin, 2023-04-30:
Thank you Patrick!
As I understand it, I can only either manually control the fan or the temperature control. When changing over, I would have to change the YAML and then flash it on the ESP. Or did I miss something, can I really switch it in HA without flashing?

I build the project for me because I want to have the temperature control in daily use and I'm super happy with it.

I'm only concerned with emergencies (hopefully only need it once in 10 years or less) that a value (HDD temp or system temp) is too high. In this case, I just want to set the fans to 100% before having to turn off the NAS for example. 

For that I searched the best solution.