patrickcollins12/esphome-fan-controller #22: Additional fans

B08Z, 2023-06-18:
Not sure this is an issue as I haven't actually started work on it. But I want to be able to connect 3 pwm fans and be able to read their rpm etc.

I have looked over the yaml and I assume I just copy the portion of code that specifies the ledc platform but change the gpio
  

If any one has a config for 2 or more fans that would be amazing.

B08Z, 2023-06-28:
> Not sure this is an issue as I haven't actually started work on it. But I want to be able to connect 3 pwm fans and be able to read their rpm etc.
>
> I have looked over the yaml and I assume I just [...]

Sorted it. Super easy. 

Just wondering can I have variable PWM outputs. So rather than all 3 fans running the same they would have independent pwm input

patrickcollins12, 2023-06-29:
Yea you can. Put the pwm on a separate gpio. You’ll need3 temp sensors too though. Copy the climate yaml to create 3 pid climates with the temp sensors and 3 ledc’s, give them all unique ids and names. home assistant will then see 3 different climate controllers.

martinhoess, 2023-07-06:
Sorry for hijacking this but my question goes in the same direction.

Is the number of fans limited in any way other than by the power supply? 
(Assuming they should all run at the same speed). 

I have 5x3 12v 80mm and 2x 12v 120mm fans as ventilation for my small office  rack. 
It would be really nice if I could control the fans dynamically via HA (depending on the office/rack temperature). Means the DHT11 I would not necessarily (only as a backup).

DunklesKaltesNichts, 2023-07-06:
the number of fans is not really limited. the number of independently controllable fans depends on the number of GPios.

 you can group all 80mm together and control the 120mm individually. it just depends on how you connect the PWM line of the fans.

martinhoess, 2023-07-07:
Thank you.

The plan was to make two units anyway (redundancy).
The 80mm are in groups of 5 (I wrote that the wrong way around). But they should all run the same speed anyway. But I will try that out soon :)

martinhoess, 2023-08-03:
I played around with it yesterday/today. Works wonderfully.

Additional fans work without problems, although I have now connected mine together as a chain and not each individually (in my case the fans should all run the same speed and they have an extra port to connect together).

The only thing I did different is, I left out the buck converter, this ESP board (not all) can be directly powered by VIN with 7-12 volts so you can use the 12V directly and the temperature sensor is powered by the 3V3 from the board. 

And the DHT11 I replaced by a DHT22 because the DHT11 has only an accuracy of +-2 degrees (the DHT11 should be banned, the price difference is so minimal that it has no reason to exist).

martinhoess, 2023-08-23:
[image]

So, here is my attempt.

3D printed enclosure. Two with DHT22 instead of DHT11 and two with Dallas DS18B20 (because the probe has a long cable and is easier to attach somewhere). And as mentioned above I left out the step down because this board can work with 12V and provides 3V for the probe. On each box there are 5x80mm fans (daisy chained). 

I still have to make the integration/display in HA nicer and also watchdog if one of the boxes should fail. The nice thing is that the fans work even if the ESP should not respond / be broken.

Thanks to @patrickcollins12 for the groundwork and creation of this project.

patrickcollins12, 2023-08-23:
[image]

This is my fan speed and cabinet climate over the last few hours. Doing it's job well.

cschillings, 2024-07-11:
How can i configure 2 fans to act like 1?
Same, RPM at same moment from the same temperature sensor.

closed 2025-04-02 by B08Z