patrickcollins12/esphome-fan-controller #24: Fritzing Img ESP Pinouts?

bfleg, 2023-10-19:
Hi @patrickcollins12  - thanks for the awesome project! This has been on my to-do list for months, although I'm intending to modify it (or at least try to!) to control speed with humidity (drying cupboard use-case!)... Although just noticed I'm also going to have the 0% PWM / fan off issue too...

However, just as I was about to get going, it seems that the Fritzing diagram of the ESP pinouts don't match my ESP (Firebeetle), is the diagram intended to be correct or is it more a rough guide and should follow the descriptions?  For example, your description and code use GPIO25, but I don't believe that's the pin shown on the fritzing?

Thanks again...

patrickcollins12, 2023-10-19:
All pins that can act as outputs can be used as PWM pins to control the fan (GPIOs 34 to 39 can’t generate PWM).

patrickcollins12, 2023-10-19:
In my experience it's very quick to test. Just play around with the pins until you get the result you want. Generally stay away from the lower pins 0-5 as they often have special purposes

patrickcollins12, 2023-10-19:
This is the canonical article

https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

bfleg, 2023-10-19:
cool resource - thanks very much for sending that on and the quick reply!