patrickcollins12/esphome-fan-controller #26: constantly reboots

Bond246, 2023-11-13:
Hello Patrick,

thanks for that repo. It is perfect for my need to cool the technic cupboard for DSL Router, NAS and these things in my livingroom.

To customize the project for my need without homeassistant but with node-red i disabled the HA API and enabled MQTT. I tried to add a 2nd temperature sensor but whatever i did it never had a value. So trying differenz GPIOs Pins, changing the sensor or add a pullup resistance it never worked. But thats not the question.

I think, in general, i havn't changed alot in your code but the scetch is never stable. Sometimes the ESP32 is running for 10 hours, sometimes only for 15min. 
The reset reason is always `Software Reset CPU`

Any ideas what the problem could be or how to debug it?

Thanks

patrickcollins12, 2023-11-14:
Does the second sensor work if you swap it into 1st position? Determine if the sensor is working first or if it is a software issue.

I haven't tried mqtt on the device but it should work... seems like an out of memory error. Can you post the logs just prior to failure?

Get the logs via:

$ esphome logs fan.yaml 

Feel free to post the yaml here too.

Bond246, 2023-11-16:
Hello patrick,

i've reduced the logs drastically to reduce the systemload which seemed to has an effect on the boot-cycle problem. So i have the same theory that either cpu or memory are being overloaded but i havn't found a way to get statistics about that from the esp32.

This is my configuration:
```yaml
substitutions:
  friendly_name: Serverschrank Fan

esphome:
  name: server-rack-fan

# Throttle writing parameters to the internal flash memory to reduce ESP memory wear / degradation
preferences:
  flash_write_interval: 15min

 
#########################
# ESP32 AND NETWORK SETUP

esp32:
  board: nodemcu-32s
  framework:
    type: arduino

# pid climate log update is noisy, dial it back to warn
logger:
  level: DEBUG
  logs: 
    climate: ERROR
    sensor: ERROR
    text_sensor: ERROR
    dht: ERROR
    pulse_counter: ERROR

debug:
  update_interval: 5s

# default HA integration, OTA updater and backup http web portal
# api:
ota:
wifi:

  # Read the wifi/pass from secrets.yaml:
  # wifi_ssid: "My Wifi XX"
  # wifi_password: "XXXXXXX"
  ssid: !secret wifi_ssid
  password: !secret wifi_password

web_server:
  port: 80

mqtt:
  broker: broker
  discovery: false
  username: !secret mqtt_user
  password: !secret mqtt_password

number:

  ## OPTIONAL:
  # RECEIVE KP, KI and KD parameters from input_text.kx helpers in 
  # Home Assistant. See the PID controller below
  # These helper values will get saved to flash thus permanently over-riding 
  # the initial values set in the PID below.

  # KP
  - platform: template
    name: kp
    icon: mdi:chart-bell-curve
[... 372 lines cut ...]
[D][text_sensor:064]: 'Device Info': Sending state '2023.10.6|Flash: 4096kB Speed:40MHz Mode:DIO|Chip: ESP32 Features:WIFI_BGN,BLE,BT, Cores:2 Revision:3|ESP-IDF: v4.4.2|EFuse MAC: C8:xx:xx:xx:xx|Reset: Software Reset CPU|Wakeup: Unknown'
[D][debug:317]: Wakeup Reason: Unknown
[D][debug:272]: Reset Reason: Software Reset CPU
[D][debug:172]: EFuse MAC: C8:C9:A3:FC:AF:D4
[D][debug:167]: ESP-IDF Version: v4.4.2
[D][debug:159]: Chip: Model=ESP32, Features=WIFI_BGN,BLE,BT, Cores=2, Revision=3
[D][debug:110]: Flash Chip: Size=4096kB Speed=40MHz Mode=DIO
[D][debug:080]: Free Heap Size: 180296 bytes
[D][debug:076]: ESPHome version 2023.10.6
[I][app:102]: ESPHome version 2023.10.6 compiled on Nov  3 2023, 11:25:15
[D][text_sensor:064]: 'Serverschrank Fan IP Address': Sending state '192.168.171.12'
[D][sensor:094]: 'Serverschrank Fan WiFi Strength': Sending state -40.00000 dBm with 0 decimals of accuracy
[I][app:062]: setup() finished successfully!
[I][mqtt:274]: MQTT Connected!
---
[D][sensor:094]: 'Luftfeuchte': Sending state 36.00000 % with 0 decimals of accuracy
[D][sensor:094]: 'Fan Speed (PWM Voltage)': Sending state 0.00000 % with 1 decimals of accuracy
[D][sensor:094]: 'Temperature': Sending state 28.10348 °C with 3 decimals of accuracy
[D][sensor:094]: 'Luftfeuchte': Sending state 36.00000 % with 0 decimals of accuracy
[D][sensor:094]: 'Fan Speed (PWM Voltage)': Sending state 0.00000 % with 1 decimals of accuracy
[D][sensor:094]: 'Temperature': Sending state 28.10387 °C with 3 decimals of accuracy
```

I made a dashed line `---` where the reboot might had been.
So there was a log of the last sensor-message and the next message is "mqtt connected" after the reboot. But my logs are provided over mqtt...

patrickcollins12, 2023-11-17:
When you say the second sensor doesn't work, what exactly are you expecting it to do? Does it log any values? Or are you expecting it to do something to the climate pid? 

The instructions for dht22 suggest that you definitely need a pull-up resistor. I haven't needed one but maybe you do, and maybe for some reason having two is what triggers needing it.

https://esphome.io/components/sensor/dht.html

Once you get two temp sensors working what will you do with the second one? Do you want an average of both sensors to drive one fan? Or will the second sensor drive a separate fan?

patrickcollins12, 2023-11-17:
Try disabling components, in particular the web server which is resource intensive. 

If you disable ota you would also free up resources. 

Start disabling some of the components I setup for debug. All of the components which send values back. They're technically only needed for debug anyway.

What kind of esp32 are you running? Different models have different ram.

patrickcollins12, 2023-11-17:
Also, is there a reason you need Mqtt rather than the HA api? The latter is more suitable and compact if the data is going to end up in HA anyway.

Bond246, 2023-11-18:
Thanks for your replay,

> When you say the second sensor doesn't work, what exactly are you expecting it to do? Does it log any values? Or are you expecting it to do something to the climate pid?
>
> The instructions for dht22 [...]

First step of integration was just to get a temperature-value. Next step would had been to use an average of both values as input for the PID. But i never get a value. Always just %NaN... I tried it with and without the pullup without any difference. The system is already integrated into its place so actually i don't have plans to add the 2nd sensor again. And in general it works well with one sensor.

My board is a DEBO JT ESP32 with Tensilica LX6 Dual-Core, 512kB SRAM and 4096MB ROM.
https://www.reichelt.de/nodemcu-esp32-wifi-und-bluetooth-modul-debo-jt-esp32-p219897.html?search=DEBO+JT+ESP32
I will try to disable some components as suggested. How can i reboot the chip when i disabled the webserver? Actually i'm doing a POST request on /switch/serverschrank_fan_esp32_restart/turn_on

> Also, is there a reason you need Mqtt rather than the HA api? The latter is more suitable and compact if the data is going to end up in HA anyway.

I don't have HA in my home-automation setup. I'm doing everything with Node-RED. All HA integration i found for Node-RED are really bad.

patrickcollins12, 2023-11-20:
Well the web server is very large and I've seen others having memory issues with it. So I suspect that is your problem. Try disabling it. If that solves the issue, maybe there is another solution for remote reboot, like mqtt.

Bond246, 2023-11-20:
Thanks so far.

I've tested another esphome package for Node-RED that uses the api. But it seems to that the connection is not stable and i get a lot of reconnects on the API.
In parallel i disabled the webinterface. Both options didn't changed anything.

So the next round:
- update all dependencies
- disable API
- disable webinterface
- disable number.platform.template: kp, ki, kd

Now i need to wait another day (or two) to see how stable it is.

Next step would be to switch from mqtt to api which means no mqtt but api. Last test was mqtt and api in parallel.

Bond246, 2023-11-20:
I've seen this option but acutally the reboots came later than 15min. So this is not the problem in general but maybe a side-effect.

If the problem stil exists with my acutal configuration i will switch from mqtt to api and will test this again.
Thanks.

DunklesKaltesNichts, 2023-11-20:
If the connection to the API is lost after 4 hours, the ESP reboots after 4 hours 15 mins.

Bond246, 2023-11-22:
So since my last changes in reducing load it seems actually much more stable.
Within the last two days there where 6 reboots. Sometimes after ~22 hourse. Sometimes already after 2 hours or 30mins.

In between i checked to use the api instead of mqtt which was an epic fail in my case. After some minutes the api connection is lost then hundreds of reconnects are following and at the end of the day i have to restart all components to get it working again.
I can only speculate the reasons for that. It is very likely that the esphome implementation for node-red is not really stable. But in some cases i thought that the esp32 is struggeling and the api connection is getting fuzzy. And further on the components try to reconnect which increases the struggles. 

So at the end of the day this will not solve my problem. 

Patrick you wrote that every platform-template section could be disabled. You mean also the sensor components or just specific template components?

Thanks so far!

patrickcollins12, 2023-12-02:
Since this seems to be an issue more with esphome memory and probably using mqtt I suggest joining the esphome discord server where they all hang out and you'll get a lot more ideas.

closed 2023-12-02 by patrickcollins12