patrickcollins12/esphome-fan-controller #27: Two Sensors

theblujuice, 2023-12-31:
Hey Patrick

I used yours as an example to build a fan temperature control for my network cabinet. Used a WT32-ETH01. 

I was thinking of using two temperature sensors. One inside and another outside. Use the one outside + x °C as the set-point. Thus doesn't matter what the room temperature it will only be a few degrees more inside the cabinet. Any thoughts?

My config at the moment based on yours:

#https://github.com/patrickcollins12/esphome-fan-controller/blob/master/console-fan.yaml

substitutions:
  device_name:  networkcabinet
  friendly_name:  "Network Cabinet"
  device_ip: !secret networkcabinet_ip

###############################################################################

esphome:
  name: ${device_name}
  platform: ESP32
  board: esp-wrover-kit

###############################################################################

preferences:
  flash_write_interval: 15min

###############################################################################

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16
  manual_ip:
    static_ip: ${device_ip}
    gateway: !secret localgateway
    subnet: !secret localsubnet
    dns1: !secret localdns1
    dns2: !secret localdns2

###############################################################################

# Enable logging
logger:

###############################################################################

# Enable Home Assistant API
api:
  encryption:
    key: !secret apikey
ota:
  password: !secret otapass

###############################################################################

# Time
time:
  - platform: homeassistant
    id: esptime

###############################################################################

i2c:
  sda: 33
  scl: 32
  scan: true
[... 232 lines cut ...]
      threshold_high: 0.4°C
      threshold_low: -1.0°C
      kp_multiplier: 0.0
      ki_multiplier: 0.04
      kd_multiplier: 0.0
      deadband_output_averaging_samples: 15
    on_state:
      - sensor.template.publish:
          id: p_term
          state: !lambda 'return -id(${device_name}_pid).get_proportional_term() * 100.0;'
      - sensor.template.publish:
          id: i_term
          state: !lambda 'return -id(${device_name}_pid).get_integral_term()* 100.0;'
      - sensor.template.publish:
          id: d_term
          state: !lambda 'return -id(${device_name}_pid).get_derivative_term()* 100.0;'
      - sensor.template.publish:
          id: o_term
          state: !lambda 'return -id(${device_name}_pid).get_output_value()* 100.0;'
      - sensor.template.publish:
          id: in_deadband_term
          state: !lambda 'return id(${device_name}_pid).in_deadband();'
      - sensor.template.publish:
          id: e_term
          state: !lambda 'return -id(${device_name}_pid).get_error_value();'

patrickcollins12, 2024-01-05:
Yep, totally possible, I like the idea. At the moment I set my temp at 30 in winter and 32 in summer. To some degree, you need to honor the max temp of your devices as well.

See this issue where I proposed how to get the average of two sensors, you could do something similar:
https://github.com/patrickcollins12/esphome-fan-controller/issues/5

nhinnenthal, 2024-05-09:
Hey,
can you update if and how you got it working? I'm essentially trying to do the same thing.
I have a chicken coop that i want to ventilate until the temperature and humidity values inside the coop match the values from outside (with a little tolerated difference).

patrickcollins12, 2024-05-09:
You can dynamically adjust the pid target temperature to be the same as your external temperature sensor. 

https://esphome.io/components/climate/#climate-control-action

Then set a dead band on your pidclimate of +/-2 degrees.

You can't control humidity with pidclimate, but you can measure it.