patrickcollins12/esphome-fan-controller #28: DHT11 temperature stops updating periodically

chrispontiga, 2024-01-04:
Thanks for sharing this project.  I recently set this up for my new network cabinet, but found that DHT11 temperature sensor stops updating its temp values periodically for 30-45 minutes at a time:

[image]

Without the temperature input, the fan speed also stays steady, then kicks up to catch up once the temperature updates again.

[image]

The logs during these states display the DHT11 returns the exact same value repeatedly, without error.

```
[00:05:14][D][sensor:094]: 'Temperature': Sending state 31.80000 °C with 3 decimals of accuracy
[00:05:14][D][sensor:094]: 'Fan Speed (PWM Voltage)': Sending state 85.48193 % with 1 decimals of accuracy
[00:05:14][D][sensor:094]: 'Humidity': Sending state 32.00000 % with 0 decimals of accuracy
[00:05:15][D][sensor:094]: 'Temperature': Sending state 31.80000 °C with 3 decimals of accuracy
[00:05:15][D][sensor:094]: 'Fan Speed (PWM Voltage)': Sending state 85.47862 % with 1 decimals of accuracy
[00:05:15][D][sensor:094]: 'Humidity': Sending state 32.00000 % with 0 decimals of accuracy
[00:05:17][D][sensor:094]: 'Temperature': Sending state 31.80000 °C with 3 decimals of accuracy
[00:05:17][D][sensor:094]: 'Fan Speed (PWM Voltage)': Sending state 85.47533 % with 1 decimals of accuracy
[00:05:17][D][sensor:094]: 'Humidity': Sending state 32.00000 % with 0 decimals of accuracy
[00:05:18][D][sensor:094]: 'Temperature': Sending state 31.80000 °C with 3 decimals of accuracy
[00:05:18][D][sensor:094]: 'Fan Speed (PWM Voltage)': Sending state 85.47204 % with 1 decimals of accuracy
[00:05:18][D][sensor:094]: 'Humidity': Sending state 32.00000 % with 0 decimals of accuracy
[00:05:19][D][sensor:094]: 'Temperature': Sending state 31.80000 °C with 3 decimals of accuracy
[00:05:19][D][sensor:094]: 'Fan Speed (PWM Voltage)': Sending state 85.46874 % with 1 decimals of accuracy
[00:05:19][D][sensor:094]: 'Humidity': Sending state 32.00000 % with 0 decimals of accuracy
[00:05:21][D][sensor:094]: 'Temperature': Sending state 31.80000 °C with 3 decimals of accuracy
[00:05:21][D][sensor:094]: 'Fan Speed (PWM Voltage)': Sending state 85.46545 % with 1 decimals of accuracy
[00:05:21][D][sensor:094]: 'Humidity': Sending state 32.00000 % with 0 decimals of accuracy
[00:05:22][D][sensor:094]: 'Temperature': Sending state 31.80000 °C with 3 decimals of accuracy
[00:05:22][D][sensor:094]: 'Fan Speed (PWM Voltage)': Sending state 85.46216 % with 1 decimals of accuracy
[00:05:22][D][sensor:094]: 'Humidity': Sending state 32.00000 % with 0 decimals of accuracy
[00:05:23][D][sensor:094]: 'Temperature': Sending state 31.80000 °C with 3 decimals of accuracy
[00:05:23][D][sensor:094]: 'Fan Speed (PWM Voltage)': Sending state 85.45885 % with 1 decimals of accuracy
[00:05:23][D][sensor:094]: 'Humidity': Sending state 32.00000 % with 0 decimals of accuracy
[00:05:24][D][sensor:094]: 'Temperature': Sending state 31.80000 °C with 3 decimals of accuracy
```

I have yet to look into the signal integrity on the data line from the DHT11.   But I have tried to swap sensors and got similar behavior.  The only other things I could think of is connecting directly to the header on the ESP rather than the 3-pin JST header, or trying shorter wires to the sensor.

Any other insights would be appreciated.

patrickcollins12, 2024-01-04:
A few theories.

I notice that the update frequency in your log is more frequent than 1 second. Note the update_frequency below. Double check your update frequency is >1.3s, since the DHT11 doesn't have a high refresh rate. 

esphome seems to be operating correctly.

Other likely issue is a faulty sensor. Since they're so inexpensive, I'd just try a new sensor. 

Maybe wiring issues. You could try adding the recommended resistor to the sensor. 

```
  - platform: dht
    pin: GPIO33
    temperature:
      name: "Temperature"
      id: console_fan_temperature
      accuracy_decimals: 3
      ...
    # the DHT11 can only be read every 1s. Use 1.3s to be safe.
    update_interval: 1.3s
```

chrispontiga, 2024-01-05:
Thanks for the reply.  The update interval is set at 1.3s from the original code.  My feeling is that it is a signal integrity issue.  I will occasionally experience warnings as follows:  

```
[18:23:14][W][component:214]: Component dht.sensor took a long time for an operation (0.06 s).
[18:23:14][W][component:215]: Components should block for at most 20-30ms.
[18:23:15][D][sensor:094]: 'Temperature': Sending state 31.79999 °C with 3 decimals of accuracy
[18:23:15][D][sensor:094]: 'Fan Speed (PWM Voltage)': Sending state 20.55629 % with 1 decimals of accuracy
[18:23:15][D][sensor:094]: 'Humidity': Sending state 33.00000 % with 0 decimals of accuracy
[18:23:16][W][component:214]: Component dht.sensor took a long time for an operation (0.05 s).
[18:23:16][W][component:215]: Components should block for at most 20-30ms.
[18:23:17][D][sensor:094]: 'Temperature': Sending state 31.79999 °C with 3 decimals of accuracy
[18:23:17][D][sensor:094]: 'Fan Speed (PWM Voltage)': Sending state 20.55299 % with 1 decimals of accuracy
[18:23:17][D][sensor:094]: 'Humidity': Sending state 33.00000 % with 0 decimals of accuracy
[18:23:17][W][component:214]: Component dht.sensor took a long time for an operation (0.05 s).
[18:23:17][W][component:215]: Components should block for at most 20-30ms.
```
I have tried two sensors already with similar results.  I do have one more from the package I can try.  I will also try a 10k pull up since that's minimal effort with some clip leads. I'll provide an update after some tinkering.

chrispontiga, 2024-01-08:
Reporting back with my troubleshooting from this weekend.  I have tried three different DHT11 sensor modules, but all appear to perform similarly where the temperature reading stops updating for some number of minutes.  The 10k pull-up resistor made no noticeable effect, because I think the pull-up may already be included on the PCB of the module I am using.  

The last DHT11 module I tried came from a different set ordered from Amazon.  It appeared the HiLetGo-brand appeared to have the highest reviews.  Using this module combined with shorter 3-inch jumpers wires (rather than the longer 6-inch wires I previous used) appeared to have an improved performance.  I'm not sure if it s a better module or the shorter wires was the factor.  But visually, the board looks identical to the no-brand-name modules I first used.  

[image]

At this point I am unsure if the non-updating value is a defect of my board, or if it's actually standard operating function of the sensor.  I would expect that the sensor to continue to provide new values since it reports 3-decimal point accuracy and it's highly unlikely that my networking cabinet is remaining to a thousandth-of-a-degree temperature stable.  But I was thinking that perhaps the DHT11 has a minimum sensitivity and requires a change of several hundredths of a degree before reporting.  I couldn't find any information to confirm with its very limited documentation.  I ordered a BME280 module for delivery tomorrow to see if performs any better.

patrickcollins12, 2024-01-09:
that graph shows it reducing to 88.2 not to 0.0?

My DHT-11

Perhaps your ESP32 board?
Is it really logging during that period or is the ESP32 locked up with memory issues during that period?

[image]

patrickcollins12, 2024-01-09:
Try reducing the update frequency to every 10 seconds. Try a better sensor like the BME280 

https://randomnerdtutorials.com/dht11-vs-dht22-vs-lm35-vs-ds18b20-vs-bme280-vs-bmp180/

Tntdruid, 2024-05-23:
Use a https://esphome.io/components/sensor/sht3xd.html for better data the dh11 is not that great.