patrickcollins12/esphome-fan-controller #32: min_power and max_power from input field

nebula7219, 2024-05-20:
Hi
How can I get min_power and max_power from an input_number helper ? I want to change the max and min power at night
Thx
Ralph

patrickcollins12, 2024-05-20:
Unfortunately, those variables can't be set or get in real time only on setup.

Esp32 time can be synced to home assistant. Then you can use the time of day to cap the number in a template output.

patrickcollins12, 2024-05-20:
Setting the time: https://esphome.io/components/time/homeassistant.html
Altering the power output: https://esphome.io/components/output/#output-set-level-action

nebula7219, 2024-05-20:
thanks for the info, that's a good solution. is it possible to use a button for restart and the new min/max_power from an input_number helper? if that's possible, can you help me with the code? sorry for my bad English, I speak German

patrickcollins12, 2024-05-21:
Yes you could try that. Try writing some lambdas that stitch those three concepts together

nebula7219, 2024-05-21:
I tried it but it doesn't work, I don't think my skills are good enough.
I tried this:
`esphome:
  name: fan_pid_controller
  on_boot: 
      then:
        - output.set_level_min_power:
          id: console_fan_speed
          level: !lambda |-
          return id(abluft_min.state) / 100.0;
`

mainmind83, 2026-10-15:
some similar here: https://github.com/patrickcollins12/esphome-fan-controller/pull/63

[image]