patrickcollins12/esphome-fan-controller #34: ESP32 Mini + box

mainmind83, 2024-06-27:
I I have created a [variant with ESP32 mini](https://github.com/mainmind83/esphome-fan-controller) and a generic power adapter for 1 fan, In case it could be interesting and improvable [[Maker World](https://makerworld.com/en/models/513745#profileId-429870)]

[image]
[image]

patrickcollins12, 2024-06-28:
Looks excellent! Love the 3d printed case. I'll see if I can put a show case on the main readme page and put this on there. What is the temperature sensor?

mainmind83, 2024-06-28:
[DHT22](https://es.aliexpress.com/item/1005004839000791.html?spm=a2g0o.order_list.order_list_main.62.21ef194dnsVDrL&gatewayAdapt=glo2esp) temperature and humidity sensor
Another external sensor could be used or attached to the box, but in my case I needed a certain distance ;)

I have updated the list of materials in the [print model](https://makerworld.com/en/models/513745#profileId-429870). The component fastening system can surely be simplified, but CAD design is not exactly my specialty (like electronics) ;D

Nuuki9, 2025-06-24:
@mainmind83 Nice case!

Did you power the ESP32 using 3.3v or the 5v VIN?  I went with 3.3v as I saw some reported issues powering this board via 5v, but havingf powered it up via the buck converter I can't get it to respond. It was fine when powered via USB so I wonder if the board blew/. The red LED is still on but can't connect to it. I'll keey troubleshooting (and may pickup another) but knowing what your pinout was would be handy, if you can share?

mainmind83, 2026-10-04:
> Looks excellent! Love the 3d printed case. I'll see if I can put a show case on the main readme page and put this on there. What is the temperature sensor?

Of course, I'm finishing new fan rack cabinet with this box 🙃

mainmind83, 2026-10-05:
## Update: the box is now part of a complete cabinet exhaust build

A follow-up to my ESP32 Mini + box build above. The controller now drives a standard 140 mm PC fan that vents warm air out of a closed office cabinet holding a small server rack.

### What's new

- **Exhaust grille:** a rectangular Voronoi grille (234 × 72 mm visible, ~78 % open area) with a transition duct from the round fan to the slot, internal baffles and walls angled 20° to steer the airflow. No visible screws: the frame latches onto sawtooth ribs on the duct and fits 18–25 mm panels. Printed in ASA.
- **Controller mount:** an optional clip holds the [Fan controller box](https://makerworld.com/es/models/513745-fan-controller-box#profileId-431322) under the duct, with a variant for the DHT22, which reads temperature and humidity at the top of the cabinet.
- **Firmware:** the improvements from #63 (RPM calibration, real fan state reporting and PID persistence) came out of this build. Some standard fans never fully stop at 0 %, so Home Assistant now shows what the fan is actually doing, not just what it was told.

### Files

- Grille, duct, clips and 1:1 cut template: [Voronoi 140mm Fan Exhaust Grille](https://makerworld.com/es/models/3398453-voronoi-140mm-fan-exhaust-grille-hidden-screws)
- Controller box: [Fan controller box](https://makerworld.com/es/models/513745-fan-controller-box#profileId-431322)

Thanks again for the project. Happy to share more details if it's useful for the README.