patrickcollins12/esphome-fan-controller #35: Improvments and toughts regarding wiring, components, temperature-sensors and fan min_power

zeus86, 2024-08-05:
First of: great Project. I was about to build basically exactly the same, but for the ventilation of a server cabinet. 

I have made some considerations in the past about the requirements for such a project, and compared it to yours, and I came across some noteworthy differences and discrepancies. Some of them could be easily implemented and would lead to improvements, others are just some suggestions, because actively temperature-regulated fans are a super-common requirement with not so many great diy-projects (like this one) that tackle this issue.

1) voltage regulator: an LM2596 can drive alot of power, and is way overkill to power just the ESP and the sensors. Most likely it will burn more power to heat as the ESP itself consumes. Much better suited would be something like this: https://www.amazon.de/gp/product/B07TTH6C3J/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1 (you will have to remove the potentionmeter-wheel-metal-part when using a fixed voltage, this unfortunately not documented). This thing is far less powerhungry also. A second one of this in conjunction with a level-shifter can also be used to drive 5v-sensors with ease, and they have a small footprint.

2) your fritzing wiring-image shows, that the TACH-singnals of multiple fans are connected to the same ESP-pin. As far as I know, this is considered bad practice. If you use more than one fan when one is not enough, the better way is just to use off-the-shelf fan-splitters, **_which always route the TACH-Pin only to the first fan_**. They should have thick enough wires to drive plenty of fans. You can easily drive 5+ 120mm fans on full blast without exceeding 1 amp (however this is a different topic for server-grade-fans. They can easily exceed 1A per Fan!). Having pulses from multiple fans on the pin will most likely confuse your hardware. In your code regarding the TACH-Readout you have defined a multiplicator-value to be able to for example "halv" the effective TACH-signal. I am relatively sure, you have the double amount of pulses, because 2 fans are hooked together where they shouldn't be. Identical fans under identical environmental conditions driven by the same PWM-signal are expected to behave the same, so you only need to read the PWM signal once (like how everyone else implements this).

3) your min_power for the fan is 13%. Most fans would stall on this value. For one, your required `min_power` is different when starting from zero compared to "actively regulating up and down" (that is the reason, why most 3d-printer-firmwares give a short 100%-burst for ~0.5secs to a fan when starting up from zero). The widely accepted _sane_ `min_pwm`-value for fans is rather something like ~25%, but this is highly depending on manufacturer and fansize. Some fans require values as high as 35% to run stable. 

4) non-pwm-fans: this could also be implemented relatively easy I think. All you need is a step-down-converter that can go up to it's own input-voltage, that you can control by an output-pin of the ESP, and hook the fans to the output of said converter instead of the 12v-rail directly. However, on 2-pin-fans you would loose your feedback-loop (the TACH-Signal), but 3-Pin Fans should work fine (for PC-Fans they are usually DC-controlled then by varying the input voltage of the fans from ~5 to 12 volts. Some Fans may require up to ~7.5V to start.). Most PWM-Fans are also capable of beeing driven by their input voltage (the datasheet sometimes explicitly describes this, for example for some fans from the manufacturer `arctic`), and in some cases it may be desirable to do so. 

5) temperature-sensors: You are using the DHT11 here. However, when you want relative humidity, this is great, but It is not really needed here, therefore I'd rather go for DS18B20-Sensors. These are usually, cheaper, more reliable, having a more exact reading and due to the fact, that these are one-wire-sensors, they use a 1wire-bus and more than one sensor can be hooked up all together to the same pin, which simplifies wiring. This would be great for applications with more than one sensor. another advantage is, that they are also available in waterproof form, which greatly increases the possible applications. The only downside (beside no humidity-reading) is the fact, that you most likely have to flash the firmware twice: the first time with a basic-config to get the IDs of the sensor and then flash again with the correct values in the real config, but this is relevant on first install only.

these are just some suggestions, but I think some of them are noteworthy or should find their way into the docs somehow, as many users in the IoT-Space are often not super-experienced with these kind of things, so sane defaults should always apply. In general I think it would greatly increase the value of this project to be a little more versatile in the first place. 

however, thanks for this awesome project, it saved me a tremendous amount of work, that's for sure. :+1: .

patrickcollins12, 2024-08-06:
Thanks, 

1. hadn't seen those smaller regulators. Do they do 3.3v too? They look good

2. Yeah that was a contribution by someone else, I don't use the tach pin. But agree it should go to different pins.

3. 13% works for my fan under all conditions, nice and short. Each fan needs its own value. Is the documentation not clear in this regard?

4. Yeah a mosfet could easily be used to pwm the whole thing with a 3pin fan. If you look through the recent comments you'll see that some folks need to use a mosfet to get the fan to stop spinning. I feel this is a much not advanced circuit and a 3-pin vs 4-pin fan is typically the same price. I'll update the documentation one day for this not advanced solution.

5. Yah needs a better temperature sensor, esphome doesn't like dht's anymore either. I generally now prefer other sensors.

Glad it worked out for you!

zeus86, 2024-08-06:
>     1. hadn't seen those smaller regulators. Do they do 3.3v too? They look good
they can be powered from 12-24v and they can do 1.8, 2.5, 3.3, 5, 9, 12 Volt using solder-bridges when removing the metal-ring from the poti or something in between using the potentiometer. this makes them perfect for everything that uses 12v as primary power-source. they are dead-cheap, small and awesome. the specs of chinese PCBs are always expected to be exaggregated, but i suppose up to 1 amp should be possible without any issues, which already gets you very far.

>     2. Yeah that was a contribution by someone else, I don't use the tach pin. But agree it should go to different pins.
I still stand by what i wrote, but i actually just looked up some docs, and having 2 countable flanks per revelation seems indeed to be expected, so a multiplier of 0.5 should always be correct. 

>     3. 13% works for my fan under all conditions, nice and short. Each fan needs its own value. Is the documentation not clear in this regard?
just testing with some old server-fans...as long as they have input voltage, they run at least with ~34% PWM regardless of what is beeing set, and they also refuse to turn off :shrug: (i am beginning to believe this is to be expected for explicit server-fans, to gurantee a minimal amount of airflow, because many PCI-cards in servers like high-speed nics and raidcontrollers would otherwise cook themself to death very quickly without external airflow). 
> 4. Yeah a mosfet could easily be used to pwm the whole thing with a 3pin fan. If you look through the recent comments you'll see that some folks need to use a mosfet to get the fan to stop spinning. [...]
:+1: 
>     5. Yah needs a better temperature sensor, esphome doesn't like dht's anymore either. I generally now prefer other sensors.
sometimes i think about just using a thermistor, because they don't block the esp-watchdog so drastically, all DHTs and Dallas-sensors i tested do unfortunately produce warnings like those:
```
[07:06:40][D][sensor:093]: 'Temperature': Sending state 28.73952 °C with 1 decimals of accuracy
[07:06:40][D][sensor:093]: 'Humidity': Sending state 46.00000 % with 0 decimals of accuracy
[07:06:40][W][component:237]: Component dht.sensor took a long time for an operation (107 ms).
[07:06:40][W][component:238]: Components should block for at most 30 ms.
```
> 
> Glad it worked out for you!
be sure it has xD

patrickcollins12, 2024-08-19:
I added a 2nd tach pin input to the console-fan.yaml for clarity.

closed 2024-08-19 by patrickcollins12