patrickcollins12/esphome-fan-controller #42: Just wondering about the pmw voltage

CuochPotatoe, 2025-01-02:
Hello i am definitely new to esp home and home assistant. Actually I am still waiting for my Pi to arrive to start assembling all of it.
But while making a plan for all of what I am planning to do I came across your project which I will definitely copy to ventilate my small gaming closet. I am going to use some Noctua NF A12 (&F12) PMW fans with my esp32. And while getting together all my data for dimensioning I noticed Noctua states that the power signal voltage for the NF A12x25 PMW fan is 5v. But you are mentioning 12 Volts in the readme; "At 100% it will be sending 12v, at 50% it will be sending 6v."
Could you explain that to me? 

Also I would like to add more sensors for air quality, outside air sensors and maybe even a soil moisture meter, probably the smt100. But I would just like these to send their data to home assistant for data logging. Is there anything I have to acknowledge with that or do I just add the code for that into the esp32 program?

Thank for charing your fantastic project in such a complete form.

mircsicz, 2025-01-02:
Your additional Sensors are a Bit behind the Score of this specific project and just overall esphome questions...

But heres an ESPhome's SMT100 Doku:
https://esphome.io/components/sensor/smt100.html

patrickcollins12, 2025-01-04:
Use a separate esp32 for the soil sensor since they're only $5.

closed 2025-01-13 by patrickcollins12