patrickcollins12/esphome-fan-controller #48: Finding the perfect PID-spot

djjoakim, 2025-04-02:
Hey!
First of all, I really love this project - this is exactly what I’ve been dreaming about for ages, so big thank you for taking your time making this!

That being said, I have some problems with the fine-tuning of my PID…
I have a pretty special solution, I have a cabinet that’s in a room where the surrounding temperature increases/decreases a lot, so I need to find a perfect PID band that goes smooth all the way, no matter if the surrounding temperature is 5c or 20c.

Soo, I have tried some different deadband settings, and KP, KI settings - and I got it pretty smooth, but then I started seeing this ”spikes” in the output value, the output value sends 0% for 2s here and there, and I can’t figure out why..
So I figure, hey - let’s try the default (aka yours) PID and deadband settings, and as you can see in the log - it started behaving more up and down after that. (My change was made 23:00).

The previous settings I had before changing to the defaults where;

KP - 0.4
KI - 0.001
KD - 0.0

Deadband low - 2.0
Deadband high - 0.5
Ki multiply - 0.01

Love to hear your thoughts about this.
Thanks so much in advance!

Ps, I’ve translated the names to Swedish so in the diagram, the colours means;

Dark blue - actual temperate
Light blue - cooling
Yellow - taget temperature
Red - just another temperature sensor in the cabinet (doesn’t affect the regulator)
Green - room temperature

Thanks again! 

[image]
[image]

kimfr, 2025-04-03:
Are you perhaps using DS18B20 temp. sensors?
In my current setup, as soon as I add 2 (or more) DS18B20 on the same one-wire - I get a similar issue. Been doing some research and found that this is actually an issue in the ESP code - which hasn't been resolved yet.

djjoakim, 2025-04-03:
> Are you perhaps using DS18B20 temp. sensors? In my current setup, as soon as I add 2 (or more) DS18B20 on the same one-wire - I get a similar issue. Been doing some research and found that this is actually an issue in the ESP code - which hasn't been resolved yet.

Yes, I have 2 DS18B20 sensors in the same one wire! Oh, this makes sense. I will try to remove the second temp sensor to see if it disappears.
Thank you so much in advance for this information!

kimfr, 2025-04-03:
I was also thinking it might be a power issue so I was going to try and power the DS18B20 from the 5V output of the ESP (according to spec you can) - just haven't had time......

djjoakim, 2025-04-08:
> I was also thinking it might be a power issue so I was going to try and power the DS18B20 from the 5V output of the ESP (according to spec you can) - just haven't had time......

I power mine through the USB 5v.

I tried removing one of the DS18B20 sensors but I keeps dropping :( maybe it’s something to do with the update time? What time do you have?

djjoakim, 2025-04-09:
> Are you perhaps using DS18B20 temp. sensors? In my current setup, as soon as I add 2 (or more) DS18B20 on the same one-wire - I get a similar issue. Been doing some research and found that this is actually an issue in the ESP code - which hasn't been resolved yet.

I think I was too fast answering my self in the last post, I still had the second sensor connected (just removed the DATA cable) and of course, it was still in the code.

So - now I have removed all the cables to the sensor and also removed it from the code and so far, it’s no spikes at all.
Thank you so much for the help!

closed 2025-04-09 by djjoakim