patrickcollins12/esphome-fan-controller #49: Fan on 100% Speed dont have the 100% RPM

bastiitsab, 2025-04-16:
Hello and thanks for this nice project! 

 I am using three of these fans:  [ARCTIC P12 PWM PST 120x120x25 ](https://www.alternate.de/ARCTIC/P12-PWM-PST-120x120x25-Geh%C3%A4usel%C3%BCfter/html/product/1856949). They are connected with the built-in Y-Cable, so only one cable goes to the ESP32.
When the fan speed PWM voltage is at 100%, the RPM displays only 1644RPM (instead of the possible 1800RPM). In initial testing with one fan, it reached nearly 1800RPM.

Is it possible to set the 100% higher to achieve more RPM?

patrickcollins12, 2025-04-17:
this could be one of a few issues:
1. that could be the max speed of the fan.
2. insufficient power
3. a software issue with the esphome fan controller.

You would need to test these indpendently.
1. rewire it to go directly into 12v rather than via PWM
2. do you have a power station to increase voltage?
3. put a multimeter between ground and the pwm out pin... it should be 3.3v when at max power.

McDive, 2025-04-22:
first of all I need to thank Mr. Collins for this project. Great, easy to implement and works fine so far. I have the same issue with the FAN RPM. But this I will ask an other time. I like to control 2 FAN's PWM(Speed) independent. But I don't know if the PID Controller can control to FAN's   ? any hint would be nice....I found this subject on Git but can't find it anymore. I think Mr Collins solved my question