patrickcollins12/esphome-fan-controller #52: While hooked up to Noctua NF-A14x25 G2 PWM Sx2-PP my ESP32 module fried

rostar99, 2025-08-09:
I am trying to cool my server closet with two 140mm Noctua on the top of the rear wall exhaust vent. I purchased these ESP32 modules https://a.co/d/8dxVeGl which include an expansion board. I powered the ESP32 board with the expansion board dcv input (which can take in 12v) and split the 12v to power the fans. I powered the DHT11 board off the 3.3v output on the expansion board. I wired everything up and poof the ESP32 module went up. Checked all the grounds and wire leads and board voltages and the second ESP32 module (with a second expansion board) went poof. I disconnected the boards and ran the fans alone and the fans work at 100% and when I ground the PWM pin they stop as expected. I noticed that when the fans are running I can measure 3.3 -2.5 vac  coming off the PWM pin. I purchased new ESP32 Freenove boards https://a.co/d/e67Bpqc  but have put have project on hold till I can figure out why the original ESP32 boards fried. Powering the ESP32 via the dc port on the expansion board works fine on the third and final board but so did the other two before the magic smoke release. Any pointers or suggestions to troubleshoot or alter the original circuit would be appreciated.

Thank you

patrickcollins12, 2025-08-09:
12v is finding its way directly into the ESP32 module and frying it. It cannot handle more than 3.6v. I haven't used that expansion board before, but assuming it works as advertised, then it is how you're wiring it together.

Would need to see a diagram or photo to help further.

rostar99, 2025-08-10:
Here is a diagram of what I tried to do with the expansion board and ESP32. I did check all the ground and power pins on the expansion board and the ESP32 module (5v 3.3v on expansion and 3.3v on ESP32 itself) and the all read correct voltage when tested with my multimeter when powered on. Only shorted when I hooked up the fans and dht11 to the pins. Additionally, I had not successfully flashed the board yet as I am having trouble getting it into download mode.
[image]

rostar99, 2025-08-11:
As a follow up I was able to get the last one into download mode but I get a TX pin error so still not programmable. I am sending them back for full replacement (glutton for punishment) to try again from scratch. I am still puzzled how just hooking up the fans, dht11 as I diagrammed and powering it up caused the release of magic smoke. I thought if the ESP32 module was not programmed the fans would run at full speed with no control?  I have emailed Noctua to confirm these particular fans can be controlled on a ESP32 PWM 3.3v output.

patrickcollins12, 2025-08-11:
I can't see anything wrong with your diagram.

It's possible although unlikely, that you've fried your board or more likely the fans and they're accidentally passing 12v back through the tach pin. I've had that problem before with fans... if they get overloaded the circuit could be shorted. Easy to test though, put a multimeter between the tach pin and the ground. Or just unplug the tach pins.

Esp32's are cheap. Don't despair. You need 30 of them anyway... for... projects.

By the way, it's possible but unlikely that the fans don't support 3.3v. They're PWM and rely on the frequency not a high voltage.

patrickcollins12, 2025-08-12:
I'd go back to basics. Plug the esp32 in directly using my original design. Just use a regular USB plug for now into the ESP32. Then use that process to whittle down the source of your error.

rostar99, 2025-08-12:
Replacement units ESP32 arrive today and I will set up powered by USB as a test set up and go from there as you suggest. I will update tomorrow with result. Thanks again.

rostar99, 2025-09-05:
When I tested one of the replacement ESP32 boards (powered by the expansion board) the expansion board shorted taking out the connected ESP32 board.  I believe my earlier problems where related to prior live testing on the expansion boards. When the expansion board fails it sends full input voltage to the ESP32 board, killing it. So the problem was caused by a combination of user error and poor failure design on the expansion board. I sucessfully set up my two noctua fans using the ESP32 Freenove board powered by a buck converter without a problem.

rostar99, 2025-09-05:
Thanks for your help and sorry about the late follow up post on the resolution.